kyleisah/Klipper-Adaptive-Meshing-Purging #158: BED_MESH_CALIBRATE generates move out of range

mrodyhin, 2023-08-17:
BED_MESH_CALIBRATE generates move out of range when variable_mesh_margin is set up and build area is close to full.
```
15:20:56
Move out of range: 15.203 -2.000 5.000 [-0.500]
15:20:56
Move out of range: 15.203 -2.000 5.000 [-0.500]
15:20:55
Move out of range: 15.203 -2.000 5.000 [-0.500]
15:20:55
99 | (15.2, 270.7) | (2.9, 297.5)
15:20:55
98 | (44.3, 270.7) | (32.0, 297.5)

....more points...

15:20:55
13 | (189.7, 28.3) | (177.4, 55.1)
15:20:55
12 | (218.8, 28.3) | (206.5, 55.1)
15:20:55
11 | (247.8, 28.3) | (235.5, 55.1)
15:20:55
10 | (276.9, 28.3) | (264.6, 55.1)
15:20:55
9 | (276.9, -2.0) | (264.6, 24.8)
15:20:55
8 | (247.8, -2.0) | (235.5, 24.8)
15:20:55
7 | (218.8, -2.0) | (206.5, 24.8)
15:20:55
6 | (189.7, -2.0) | (177.4, 24.8)
15:20:55
5 | (160.6, -2.0) | (148.3, 24.8)
15:20:55
4 | (131.5, -2.0) | (119.2, 24.8)
15:20:55
3 | (102.4, -2.0) | (90.1, 24.8)
15:20:55
2 | (73.4, -2.0) | (61.1, 24.8)
15:20:55
1 | (44.3, -2.0) | (32.0, 24.8)
15:20:55
0 | (15.2, -2.0) | (2.9, 24.8)
15:20:55
bed_mesh: generated points
Index | Tool Adjusted | Probe
15:20:55
Generating new points...
```

mrodyhin, 2023-08-17:
Here is some more debug info:
```
17:18:25
2 | (73.4, -2.0) | (61.1, 24.8)
17:18:25
1 | (44.3, -2.0) | (32.0, 24.8)
17:18:25
0 | (15.2, -2.0) | (2.9, 24.8)
17:18:25
bed_mesh: generated points
Index | Tool Adjusted | Probe
17:18:25
Generating new points...
17:18:25
KAMP adjustments successful. Happy KAMPing!
17:18:25
Adapted mesh bounds: (2.903, 24.8), (264.7, 297.524).
17:18:25
Fuzz amount is 0, mesh points not fuzzed.
17:18:25
Mesh margin is 2.0, mesh bounds extended by 2.0mm.
17:18:25
Default mesh bounds: (-3.3, 26.8), (262.7, 303.8).
17:18:25
Adapted probe count: 10,10.
17:18:25
Default probe count: 10,10.
17:18:25
Algorithm: bicubic.
```
My print area size is 275x300

And here is my probe config:
```
##########################################
# # Probe
##########################################

[bltouch]
sensor_pin: ^toolhead:PB8
control_pin: toolhead:PB9
stow_on_each_sample: false
probe_with_touch_mode: true
x_offset: -12.3
y_offset: 26.8
#z_offset: 1.460
speed: 30

[bed_mesh]
speed: 100
horizontal_move_z: 5
mesh_min: -3.3, 26.8
mesh_max: 262.7, 303.8
probe_count: 10, 10
algorithm: bicubic
```

plugowski, 2023-08-19:
Oh yeah, I faced the same issue...

kyleisah, 2023-08-20:
This should now be fixed, please update. :)

closed 2023-08-20 by kyleisah

Bart97, 2023-09-01:
I don't think this is fixed yet. I've updated to the latest version, but I'm still seeing a similar issue:

```
09:33:10
Move out of range: 54.559 -0.671 10.000 [5.000]
09:33:10
Move out of range: 54.559 -0.671 10.000 [5.000]
09:33:10
Move out of range: 54.559 -0.671 10.000 [5.000]
09:33:10
bed_mesh: zero_reference_position is (125.00, 125.00)
09:33:10
19 | (195.4, 178.2) | (195.4, 198.0)
09:33:10
18 | (148.5, 178.2) | (148.5, 198.0)
09:33:09
17 | (101.5, 178.2) | (101.5, 198.0)
09:33:09
16 | (54.6, 178.2) | (54.6, 198.0)
09:33:09
15 | (54.6, 133.5) | (54.6, 153.3)
09:33:09
14 | (101.5, 133.5) | (101.5, 153.3)
09:33:09
13 | (148.5, 133.5) | (148.5, 153.3)
09:33:09
12 | (195.4, 133.5) | (195.4, 153.3)
09:33:09
11 | (195.4, 88.8) | (195.4, 108.5)
09:33:09
10 | (148.5, 88.8) | (148.5, 108.5)
09:33:08
9 | (101.5, 88.8) | (101.5, 108.5)
09:33:08
8 | (54.6, 88.8) | (54.6, 108.5)
09:33:08
7 | (54.6, 44.1) | (54.6, 63.8)
09:33:08
6 | (101.5, 44.1) | (101.5, 63.8)
09:33:08
5 | (148.5, 44.1) | (148.5, 63.8)
09:33:08
4 | (195.4, 44.1) | (195.4, 63.8)
09:33:08
3 | (195.4, -0.7) | (195.4, 19.1)
09:33:08
2 | (148.5, -0.7) | (148.5, 19.1)
09:33:08
1 | (101.5, -0.7) | (101.5, 19.1)
09:33:08
0 | (54.6, -0.7) | (54.6, 19.1)
09:33:07
bed_mesh: generated points
Index | Tool Adjusted | Probe
09:33:07
Generating new points...
09:33:07
probe: open
09:33:05
Attaching Probe
09:33:04
KAMP adjustments successful. Happy KAMPing!
09:33:04
Adapted mesh bounds: (54.559, 19.079), (195.441, 198.0).
09:33:04
Fuzz amount is 0, mesh points not fuzzed.
09:33:04
Mesh margin is 0, margin not increased.
09:33:04
Default mesh bounds: (35.0, 10.0), (240.0, 198.0).
09:33:04
Adapted probe count: 4,5.
09:33:04
Default probe count: 5,5.
09:33:04
Algorithm: lagrange.
```

I've attached the gcode file with the issue:
[ABS_Merged-Esun ABS+ Black.gcode.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12495534/ABS_Merged-Esun.ABS%2B.Black.gcode.txt)

kyleisah, 2023-09-02:
I had a feeling something was going on with your config. The bed_mesh_calibrate macro is made so it will never exceed what you have defined in your config. So that makes sense.