kyleisah/Klipper-Adaptive-Meshing-Purging #159: How can I change the adaptive purge settings?

SXRguyinMA, 2023-08-19:
How can I change the adaptive purge settings?

The KAMP config file is read-only and I can't see how to make it editable so I can change the settings

kyleisah, 2023-08-20:
In the same directory as your printer.cfg file, there should be a KAMP_Settings.cfg file, that is where you're supposed to change the settings for all of KAMP.

Do you have that file there?

[image]

SXRguyinMA, 2023-08-20:
No but I have Line_Purge.cfg however changing the settings in there doesn't change anything as far as the output is concerned

[image]

kyleisah, 2023-08-20:
I think you should go back through the instructions. You don't need to move anything out of the KAMP folder. You just need the settings file, and make your changes in there.

closed 2023-08-21 by kyleisah

SXRguyinMA, 2023-08-22:
> I think you should go back through the instructions. You don't need to move anything out of the KAMP folder. You just need the settings file, and make your changes in there.

Alright I'll look into that, thanks. I don't recall moving anything when I did the initial install but I'll make sure I don't.