kyleisah/Klipper-Adaptive-Meshing-Purging #165: Kamp is not meshing

Awesome4all, 2023-08-26:
I have been trying at this for the past 2 hours. when I start my print Kamp loads but does not create the mesh around my part it just does the default mes. Yes I have Tried the M117.

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
I have tried changing mesh size in printer.cfg. 

**Screenshots**
[image]
[image]

**log and gcode**
[kamp issue.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12444641/kamp.issue.zip)

kyleisah, 2023-08-26:
Are you putting M117 before print start *in your slicer*? The M117 trick doesn't work from inside the print start macro, it has to be in the slicer's gcode before PRINT_START is called

kyleisah, 2023-08-26:
Like this:
[image]

Awesome4all, 2023-08-26:
oh okay let me try that

Awesome4all, 2023-08-26:
Yes that worked tysm

closed 2023-08-26 by Awesome4all

hapsti, 2023-08-29:
what is the M117 trick?

kyleisah, 2023-08-29:
> what is the M117 trick?

Have a look in the troubleshooting section in the readme. 

It's a trick to force moonraker to define objects in the correct spot in your gcode file.

Awesome4all, 2023-08-29:
> what is the M117 trick?

Yes it was