kyleisah/Klipper-Adaptive-Meshing-Purging #166: Printing in air after meshing

reboruinz, 2023-08-29:
**Describe the issue**
- If I add BED_MESH_CALIBRATE anywhere in my start g code it creates the adaptive mesh as expected, however the nozzle starts printing ~2mm higher than it's supposed to. As an example, I print on 0.6 nozzle so layer 1 = 0.35 mm however layer 1 starts printing at 1 or 2 mm.

- Voron 2.4r2 300mm with stealthburner, inductive probe, and bed probe behind bed (stock LDO kit format)
- Using adaptive mesh and purge
- Only parameters changing are volumetric flow rate 12 to 20

**What slicer are you using?**
Orcaslicer and Prusaslicer

**Troubleshooting**
- Reinstalled Klipper on motherboard and Pi. Issue fixed itself for several days and then started happening again. 
- Tried more or less the same config on an ender 3 v2 and everything works fine.
- Reset z offset several times.
- Confirmed manually calibrating z and manually moving to 0 is accurate.
- Sliced multiple files

**Screenshots**
[image]

**Klipper log file:**
[klippy (1).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12463085/klippy.1.log)

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
I have tried to re order and omit many things from this start code and unless I straight up don't mesh, I keep having the issue:

[gcode_macro PRINT_START]
gcode:
    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}
    {% set chambertemp = params.CHAMBER|default(0)|int %}
    
    STATUS_HEATING                         ; heating LED status
    M190 S{bedtemp}                        ; set and wait for final bed temp
    M109 S150                              ; set and wait for nozzle to reach 150

    G90                                    ; absolute positioning
    BED_MESH_CLEAR                         ; clear any bed mesh
    STATUS_HOMING                          ; homing LED status
    G28                                    ; home all axes
    
    STATUS_LEVELING                        ; leveling LED status
    QUAD_GANTRY_LEVEL                      ; level the gantry
    G28 Z
   
    STATUS_MESHING                         ; meshing LED status
    BED_MESH_CALIBRATE                     ; create bed mesh
    
    STATUS_BUSY                            ; busy LED status
    G0 X300 Y306.50 Z20 F3000              ; position nozzle at back right
    
    STATUS_HEATING                         ; heating LED status
    M109 S{hotendtemp}                     ; set and wait for final nozzle temp
    
    STATUS_CALIBRATING_Z                   ; z calibration LED status
    G28 Z                                  ; perform final z home
    
    STATUS_BUSY                            ; busy LED status
    LINE_PURGE                             ; KAMP purge
    STATUS_PRINTING                        ; printing LED status

######### Endstop and mesh from last print #############

#*# [stepper_z]
#*# position_endstop = -0.880
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  1.009906, 1.011156, 1.019906, 1.024906, 1.008656
#*# 	  0.958656, 0.959906, 0.973656, 0.969906, 0.941156
#*# 	  0.957406, 0.963656, 0.968656, 0.966156, 0.944906
#*# 	  0.962406, 0.962406, 0.971156, 0.974906, 0.956156
#*# 	  0.997406, 0.989906, 0.999906, 1.008656, 0.991156
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*# min_x = 40.0
#*# max_x = 260.0
#*# min_y = 40.0
#*# max_y = 260.0

Thank you! I LOVE this feature on my other machines :)

stahlfabrik, 2023-08-29:
You should imho not home z after bed mesh. I bet that is your problem

reboruinz, 2023-08-29:
> You should imho not home z after bed mesh. I bet that is your problem
> 
> you do that after gql. But not after bed mesh

Thanks for the feedback. I can try later today, but I can confirm that I have z home after mesh on my ender 3 with bl touch and that works fine.

reboruinz, 2023-08-30:
@kyleisah thanks for the feedback! unfortunately I am still having issues. I have manually isolated it to the mesh process itself by doing the following:

1) Heat entire system
2) G28
3) Quad Gantry Level
4) G28
5) Move Z to 0.35, confirm height is correct
6) BED_MESH_CALIBRATE 
7) Move Z to 0.35, confirm height is now 2 mm higher than it's supposed to

I also tested with a real print which also failed again:
[gcode_macro PRINT_START]
gcode:
    {% set bedtemp = params.BED|int %}
    {% set hotendtemp = params.HOTEND|int %}
    {% set chambertemp = params.CHAMBER|default(0)|int %}

## CLEAR MESH AND INITIAL HOME ##
    BED_MESH_CLEAR                         ; clear any bed mesh
    G90                                    ; absolute positioning
    STATUS_HOMING                          ; homing LED status
    G28                                    ; home all axes

## MOVE NOZZLE TO CENTER FOR BED HEATING ##
    STATUS_BUSY                            ; busy LED status
    G0 X150 Y150 Z100 F3000                ; position nozzle at center for heating

## WAIT FOR BED TOO REACH TEMP ##
    STATUS_HEATING                         ; heating LED status
    M190 S{bedtemp}                        ; set and wait for final bed temp
    M109 S150                              ; set and wait for nozzle to reach 150

## RUN QUAD GANTRY LEVEL ##
    STATUS_LEVELING                        ; leveling LED status
    QUAD_GANTRY_LEVEL                      ; level the gantry

## XYZ HOME ##   
    STATUS_HOMING                          ; homing LED status
    G28                                    ; home all axes

## CREATE MESH ##
    STATUS_MESHING                         ; meshing LED status
    BED_MESH_CALIBRATE                     ; create bed mesh

## MOVE NOZZLE TO BACK RIGHT FOR FINAL HEATING ##    
    STATUS_BUSY                            ; busy LED status
    G0 X300 Y306.50 Z20 F3000              ; position nozzle at back right

## WAIT FOR NOZZLE TO COME TO TEMP ##
    STATUS_HEATING                         ; heating LED status
    M109 S{hotendtemp}                     ; set and wait for final nozzle temp
      
## PURGE AND PRINT ##    
    STATUS_BUSY                            ; busy LED status
    LINE_PURGE                             ; KAMP purge
    STATUS_PRINTING

reboruinz, 2023-08-30:
jesus christ.. figured it out i think..

bed_mesh was missing "zero_reference_position: 150,150 #for use with stock z endstop"

closed 2023-08-31 by kyleisah