kyleisah/Klipper-Adaptive-Meshing-Purging #167: preprocess_cancellation v0.2.0 adding extra EXCLUDE_OBJECT_DEFINE NAME=ENDGCODE

macsproto, 2023-08-31:
Hello,

When slicing a single part on Superslicer I get the following error when Adaptive Mesh is attempted:

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

Looking at the gcode file itself, it looks like the preprocess cancellation has added an extra line:
; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 2 known objects
EXCLUDE_OBJECT_DEFINE NAME=CatchDrainValveMount_obj_id_0_copy_0 CENTER=184.962,170.381 POLYGON=[[147.454,132.956],[147.454,195.502],[210.002,195.502],[210.002,132.956]]
EXCLUDE_OBJECT_DEFINE NAME=ENDGCODE

Somehow it is seeing two objects when there is only one in the slicer.

Removing the extra line resolves the problem and the print is able to go through adaptive mesh.

What inside of Mainsail is causing the extra EXCLUDE_OBJECT line to be written when files are uploaded?

Thanks in advance for your help.
[CatchDrainValveMount.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12483377/CatchDrainValveMount.zip)

kyleisah, 2023-08-31:
I don't think this is a KAMP issue as KAMP is not *causing* this problem. I'll take a look at it, but I can't promise anything unfortunately.

kyleisah, 2023-09-13:
Is this still happening for you? I'm looking at the gcode and I've no clue why the preprocessor just made an object called ENDGCODE for no reason lol.

kyleisah, 2023-09-13:
I think this block in your gcode might be confusing it: 
[image]

No clue why that's happening though.

macsproto, 2023-09-13:
Yeah it’s still bugging me but I’ve kinda given up on it since I had to
address some canbus issues.  Just have to live with pleb mesh calibrate for
now until I find some downtime.

Thanks for looking into this again!

closed 2023-09-17 by kyleisah