kyleisah/Klipper-Adaptive-Meshing-Purging #168: Mesh extends outside of the fade range

pireybr, 2023-09-05:
**Describe the issue**
i'm getting these error when I try to print some large files

bed_mesh: setting zero reference at (175.00, 175.00, -0.015000)
bed_mesh:  Mesh extends outside of the fade range, please see the fade_start and fade_end options inexample-extras.cfg. fade distance: 1.35 mesh min: -0.1528mesh max: 5.2487
bed_mesh:  Mesh extends outside of the fade range, please see the fade_start and fade_end options inexample-extras.cfg. fade distance: 1.35 mesh min: -0.1528mesh max: 5.2487
bed_mesh:  Mesh extends outside of the fade range, please see the fade_start and fade_end options inexample-extras.cfg. fade distance: 1.35 mesh min: -0.1528mesh max: 5.2487

**What slicer are you using?**
Superslicer

**Troubleshooting**
What troubleshooting steps have you tried?

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-09-06:
Please share a Klipper log file.

pireybr, 2023-09-11:
[klippy (9).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12571475/klippy.9.log)

kyleisah, 2023-09-11:
I think if you adjust your fade values in [bed_mesh] this error will clear up. Seems like the fade distance you're using is maybe too short and klipper doesn't like it. 

Have a look here for default values:

https://www.klipper3d.org/Bed_Mesh.html?h=fade#mesh-fade

closed 2023-09-17 by kyleisah

pireybr, 2023-09-17:
thanks, it's now working