kyleisah/Klipper-Adaptive-Meshing-Purging #171: Error activating KAMP jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'exclude_object'

Nightkurz, 2023-09-12:
I have not been able to activate KAMP Adaptive meshing, an error is issued every time the command "BED_MESH_CALIBRATE" is executed.

I've been using PrusaSlicer 2.6.1

I've read every troubleshoot post i've found, including adding the M117 command on top of the PRINT_START command on the slicer, I've folowed the PRINT_START recomendations Jonteck2.

This is the console output of the error:

[image]

Here is tle klipper log file and the gcode of my last failed try

[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12591538/klippy.log)

[First_Layer_Test_11m_0.20mm_200C_PLA_ENDER3V2NEO.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12591542/First_Layer_Test_11m_0.20mm_200C_PLA_ENDER3V2NEO.zip)

If you need anything else to pinpoint the problem just let me know.

Thanks in advance.

kyleisah, 2023-09-13:
I don't see `[exclude_object]` defined in your config.

kyleisah, 2023-09-13:
https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#requirements

Nightkurz, 2023-09-13:
You're right that was missing somehow, however now any print that i try just makes the full mesh even a small xyz cube makes a full mesh across the bed.

[image]

this is the gcode prewiew and file,

[image]

[xyzCalibration_cube_24m_0.20mm_200C_PLA_ENDER3V2NEO.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12600010/xyzCalibration_cube_24m_0.20mm_200C_PLA_ENDER3V2NEO.zip)

any advice on that?

Thanks in advance!

closed 2023-09-14 by Nightkurz

kyleisah, 2023-09-17:
https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#troubleshooting