kyleisah/Klipper-Adaptive-Meshing-Purging #176: smart park not going to purge start

TomW1605, 2023-09-18:
**Describe the issue**
in a recent release you made it so the smart park would go to the start coordinates of the purge line (#126). this doesn't seem to be working. I'm guessing the purge logic has changed and the park logic hasn't been updated to match

this is the console output im getting:
```
13:57:05 // setting park to start of purge
13:57:05 // Smart Park location: 71.35,79.85.
13:58:22 // Moving filament tip 0.0mms
13:58:22 // KAMP purge is not using firmware retraction, it is recommended to configure it.
13:58:22 // KAMP purge starting at 95.0, 64.85 and purging 30.0mm of filament, requested flow rate is 12.0mm/s3.
```

i added the print on the first line to make sure it is actually getting into the if statement.

**What slicer are you using?**
Superslicer

**Troubleshooting**
What troubleshooting steps have you tried?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
[light sensor cover v1_0.2mm.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12645067/light.sensor.cover.v1_0.2mm.txt)
this is slightly modified, i set the bed temp to 0 and added a pause before the start of the actual print for debugging

start macro:
```
[gcode_macro START_PRINT]
gcode:
    {% if 'HOTEND_TEMP' in params|upper %}
        {% set hotend_temp = params.HOTEND_TEMP|int %}
    {% else %}
        { action_raise_error('Must provide HOTEND_TEMP parameter') }
    {% endif %}
	
    {% if 'BED_TEMP' in params|upper %}
        {% set bed_temp = params.BED_TEMP|int %}
    {% else %}
        { action_raise_error('Must provide BED_TEMP parameter') }
    {% endif %}
	
    {% if 'TOTAL_LAYERS' in params|upper %}
        SET_PRINT_STATS_INFO TOTAL_LAYER={ params.TOTAL_LAYERS|int }
    {% endif %}
	
    {% set initial_hotend_temp = (hotend_temp / 1.4)|round(0)|int %}
    
    LED_ON
    
    M140 S{bed_temp} ; set bed temp
    
    M104 S{initial_hotend_temp} ;start heating hotend WITHOUT WAIT
    
    M190 S{bed_temp} ; wait for bed temp
    
    G28;
    
    BED_MESH_CALIBRATE
    
    SMART_PARK
    
    M109 S{hotend_temp} ; wait for extruder temp
    
    MR_NOTIFY TITLE=$printer_name MESSAGE="Heating finished, starting print"
    
    LINE_PURGE
```

**Additional context**
Add any other context about the problem here.

closed 2023-09-19 by kyleisah

TomW1605, 2023-09-20:
i see a few ways that this could be acheived.

1. add a parameter to the smart park macro that tells it what purge your using. if its not there then do what it does now.
2. add a macro to each purge that is a purge spesific smart park.

for both of these it might be worth adding a macro to each purge that does the calculations for the start point so that you are not doing that calculation in 2 places and when it is updated you dont need to update it twice.

i think the advantage of having the park at the start of the purge and not having extra blobs on the bed or ooze hanging from the extruder is worth adding this. im happy to take a stab at this if your open to a PR (if not ill just leave the changes local)

jonathanoeijoeng, 2023-09-20:
I have same experience when smart park is different with purge start location. My concern is blob from extruder can fall in to print area since it start to increase hot end in smart park area. Would be useful if we can set smart park location same with start purge location. Vote for TomW1605

kieraneglin, 2023-09-27:
@kyleisah you should be able to definitely know the start position of the purge by splitting out the initial move into its own macro that is referenced by both the purge and smart park. It's outdated and I'm sure it could be done better, you can see an example of this in my PR #128

kyleisah, 2023-09-27:
I will look into it but I cannot promise anything. I'm not comfortable with adding more macros to *adjust* functionality, but maybe there's another way this can be done using some gcode macro variable juggling.

jonathanoeijoeng, 2023-09-28:
I have changed as mentioned here and works well. Smart Park position exactly where line purge will be done.
I just add action_respond_info script in macro _MOVE_TO_LINE_PURGE_START to show smart park position in console fluidd.

Hopefully it can be merged to main branch.

-Jonathan-

> @kyleisah you should be able to definitely know the start position of the purge by splitting out the initial move into its own macro that is referenced by both the purge and smart park. It's outdated and I'm sure it could be done better, you can see an example of this in my PR #128