kyleisah/Klipper-Adaptive-Meshing-Purging #179: KAMP doesn't work with OctoKlipper

jimmysdiy, 2023-09-20:
I am using OctoKlipper using OctoPrint 1.0 w/ Python 3.9. Installed KAMP and all configured. KAMP doesn't seem to work. However, with the same Setup on Mainsail, it works beautifully. I also tested using same config files on my duplicate machine running mainsail and it works

Seems like KAMP has no effect. During the beginning of the print, the printer seem to perform full bed mesh and purge is standard side purge instead of goopy purge in the front

Slicer - 
PrusaSlicer 2.6

**Troubleshooting**
Used Same settings & Cfg on mainsail and it works

**Screenshots**
n/a - I can record a video and provide if required

**Klipper log file:**
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12675397/klippy.log)

**Sliced Gcode**
I had to add txt extension to get gcode file attached. remove .txt to read gcode
[square_tower_54m_0.30mm_200C_PLA_ENDER3PRO.gcode.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12675409/square_tower_54m_0.30mm_200C_PLA_ENDER3PRO.gcode.txt)

**Additional context**
n/a

kyleisah, 2023-09-20:
Oof, I know nothing about octoklipper or how that all works with stuff like this. But all KAMP is, is a set of macros so I'm not sure why it wouldn't work. I might have to dig into this a bit.

kyleisah, 2023-09-20:
Well, after looking at your files here, I have some notes.

It looks like [exclude_object] is not working correctly. Either you have it in your config but object processing is not `true` in moonraker.conf, or the object processor is just not doing that it's supposed to be doing. 

The "side purge" is actually not standard, and only happens when no objects are found, or the bed is full. The purge in front of the print area is actually the intended default.

I'm also not seeing your slicer output labeled objects, So that could be an avenue as well.

I don't think this is a KAMP or OctoKlipper issue, I think you just don't have it set up right.

jimmysdiy, 2023-09-20:
I'll take a look at it. I literally imported the same configs from my other
machine.

I do not have mainsail.cfg given I am using OctoPrint.

I did add full path for the cfg after I added this as a bug.

Oddly enough when I disabled smart park and line purge. It purged
vertically at the print spot. Weird.

julianschill, 2023-09-22:
The thing is that when using Octoklipper there is no moonraker doing the preprocessing. So you have to do it in the slicer. Check [here](https://github.com/kageurufu/preprocess_cancellation) for how to set it up.

jimmysdiy, 2023-09-24:
I think you may have hit the jackpot. But i gave up on OctoPrint w/ Klipper. I feel the integration is horrible, I'll make a switch to Mainsail!

closed 2023-09-26 by kyleisah

TheLexus, 2023-10-13:
Just a side note... with PrusaSclicer it works because the order of gcode is different... sad. I use Fluidd with Klipper, and a switch in the slicer is enough for me... :) Fluidd works quite well, but I have no experiences with others. Maybe in the future I try Mainsail!