kyleisah/Klipper-Adaptive-Meshing-Purging #180: KAMP comand ln does not work for linux pc with two klipper printers

Marcho1324, 2023-09-20:
Hi! I am running two klipper printers on a linux laptop one with cp210 drivers the other one with ch34 drivers to serial 
So i was trying to run the commands for the  installation in the cmd from the laptop
everything was fine until the ln -s command
this is the output 

marin@marin-ThinkPad-X200s:~$ cd
marin@marin-ThinkPad-X200s:~$ git clone https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging.git
fatal: destination path 'Klipper-Adaptive-Meshing-Purging' already exists and is not an empty directory.
marin@marin-ThinkPad-X200s:~$ ln -s ~/Klipper-Adaptive-Meshing-Purging/Configuration printer_data/config/KAMP
ln: failed to create symbolic link 'printer_data/config/KAMP': No such file or directory
marin@marin-ThinkPad-X200s:~$

everything is updated

kyleisah, 2023-09-20:
I think you'll need to adjust the commands so the symlink happens in the right directories. I'm guessing it's not just `printer_data` for you. It's probably something like `printer1_data` or `printer2_data` or whatever you named those instances.

If you do `cd` and do `ls` you should be able to see the directories.

kyleisah, 2023-09-20:
[image]
In my case I would need to do the symlink like this:
`ln -s /Klipper-Adaptive-Meshing-Purging/Configuration BoxZero_data/config/KAMP`

Marcho1324, 2023-09-20:
ok it worked thanks 
yes for me it was n and b default values

kyleisah, 2023-09-20:
Not a lot of people do the multi-printer setup lol, I'm glad you got it working. Happy printing!

Marcho1324, 2023-09-20:
it was a pain in the ass with the drivers for cp210 and ch34