kyleisah/Klipper-Adaptive-Meshing-Purging #181: The purge line ends up in the middle of the parts for printers where X,Y (0,0) is the center of the bed

duembeg, 2023-09-21:
https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/blob/c7b4b45ecb6b4f1b9b1f16dd5f7f425b9198a6f3/Configuration/Line_Purge.cfg#L33-L38

The purge line ends up in the middle of the parts for printers where X,Y (0,0) is the center of the bed, I belive one can use "printer.toolhead.axis_minimum.x" and "printer.toolhead.axis_minimum.y" instead of comparing to zero (0) 

Thoughts?

kyleisah, 2023-09-21:
So, a delta printer?

kyleisah, 2023-09-21:
This is interesting. I’ll have to look into it.

duembeg, 2023-09-22:
> So, a delta printer?

Not delta, I have printer with different size beds; I have always found it easier to have 0,0 at center. I have seen other configure it this way too. I don't have  printers with 0,0 at the corners, but to get me going I changed the code so it doesn't compare to zero but right now it also doesn't check for a minimum not to go off the bed, I will try using the printer.toolhead.axis... but of course it should work on both scenarios. But loving KAMP :)

barza-k, 2023-09-22:
Yeah same here. Using a Delta Printer Purge Line is always in the middle (0 , 0) even inside objects.

kyleisah, 2023-10-07:
Just letting you know I haven't forgotten this. Looking over issues and PRs right now and I'm seeing some small things I overlooked. I've already got the parking fixed the way it should be, looks like some of that same calculation should be fixed for the purge locations. 

I'll be working on this as my schedule allows!

kyleisah, 2023-11-02:
I don't know if there's much I can do about this. I developed this to be used with typically/correctly configured machines that are CoreXY and Cartesian. Delta's have never been supported because I don't have a delta to develop/test on. 

The way you're supposed to configure your machine is to have the front left of the bed be 0,0. Anything other than that is either incorrect configuration, or abnormal configuration.

The problem is, I could make the purge line compare 0 and axis minimums, but having the minimum position rather than zero be compared may allow the purge to happen off of the bed altogether. 

The difficulty isn't in solving the problem, it's making it work for everyone. 

So with that, I think it's best you just run your own fork/version of KAMP since you're used to having the origin be at the center of the bed.

closed 2023-11-02 by kyleisah