kyleisah/Klipper-Adaptive-Meshing-Purging #182: UndefinedError: 'dict object' has no attribute 'polygon'

lsmelvin, 2023-09-22:
**Describe the issue**
A clear and concise description of what the bug is...

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

**What slicer are you using?**
Prusaslicer, Superslicer, Cura, Orcaslicer...

superslicer

**Troubleshooting**
What troubleshooting steps have you tried?

looking online and triple checking work

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Klipper log file:**
Please add klipper log file here...

[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12697702/klippy.log)

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

already added exclude_object in printer.cfg and checked label objects in superslicer followed setup guides and videos to a T so im unsure what im doing wrong.

[image]

[image]

julianschill, 2023-09-22:
Please add a sliced gcode file. That helps to find the problem.

Theo9909, 2023-09-24:
Recently had to rebuild my system. Now this error pops up everytime I try using adaptive meshing. I had no issues until the redo. Used the same macros, and printer.cfg files. 

 How can I include the preprocessed gcode?

Theo9909, 2023-09-24:
[gcodes-202380-181521.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12709774/gcodes-202380-181521.zip)
Gcode file from klipper setup attached.

julianschill, 2023-09-24:
Something went wrong with the preprocessing. Please also add the moonraker.log

Theo9909, 2023-09-24:
[moonraker.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12709908/moonraker.zip)

julianschill, 2023-09-25:
You are using "machinekit" as gcode flavor in Prusaslicer. This results in wrong G1 commands with Axx instead of Exx, which causes the problem.
Change the gcode flavor to "Klipper" (or "Marlin legacy" for older Prusaslicer versions) in the printer settings and it should work.

Theo9909, 2023-09-25:
That's interesting, as I haven't changed that setting in prusa slicer since before I had the Ethernet issues. I'll get that changed and what it work like it always did.

Thank you, I will update shortly.

Theo9909, 2023-09-26:
That did it. Not sure how that changed, but two days of trying everything and never looking at that setting once against n proves that I need to look at the simple things first. Thank you for your help! Hopefully the OP is able to gleam some info off of this. Take care.

kyleisah, 2023-09-26:
> That did it. Not sure how that changed, but two days of trying everything and never looking at that setting once against n proves that I need to look at the simple things first. Thank you for your help! Hopefully the OP is able to gleam some info off of this. Take care. 

I would have never thought to check gcode flavor. I will be sure to make a note of this or add something in the troubleshooting section for it.

closed 2023-09-26 by kyleisah