kyleisah/Klipper-Adaptive-Meshing-Purging #185: Añadir posición de la purga personalizada

arsonists, 2023-09-25:
First of all thank you for your work, it is really good.

It is not a failure as such, but depending on how you look at it, it can be.

Previously it allowed to select X and Y coordinates and direction for the purge and it was very useful to select an area "outside the printing area", (putting negative values to the coordinates) for example in the prusa sheets.

Now I usually have failures when the parts cover the whole printing area, because it has to move "outside" to print.

I hope I am explaining myself, but I would find it very interesting to add again the option of being able to put the purge where you want. Otherwise in my case, it is better for me to go back to the previous version.

Best regards.

dog2bert, 2023-09-26:
Can you run only the purge macro or do you have to "install" everything?

arsonists, 2023-09-26:
It is not possible, since the configuration is now unified for both the purge line and the mesh.

kyleisah, 2023-09-26:
> Can you run only the purge macro or do you have to "install" everything?

You can only use the purging if you wish, that's what I do on my machines with no probe. Just don't include the Adaptive_Meshing.cfg in KAMP_settings.

kyleisah, 2023-09-26:
> First of all thank you for your work, it is really good.
>
>
>
> It is not a failure as such, but depending on how you look at it, it can be.
>
>
>
> Previously it allowed to select X and Y coordinates and [...]

Defining a place to purge wouldn't be very adaptive though. 

If you're having issues with out of bounds movements, you should check your config/slicer settings to make sure that 0,0 is defined correctly on your machine. By design, the line purge cannot exceed your print area because if the bed is full of objects, the purge will move to X0 and purge along Y.

arsonists, 2023-09-26:
HHello again.

Maybe I didn't explain myself well.

My bed of is bigger than the area I have defined for printing (0,0) and right now, with your previous version I make the puga outside (0,-5).

I send you a picture and you will understand. That is why I find it convenient to be able to define where you want the purge line.

https://prnt.sc/t7yzSyrNUNey

kyleisah, 2023-09-26:
You can always just *not* use the adaptive purging and have your own purge macro. You can keep the adaptive meshing and remove the adaptive purging. Seems like you might be best off that way.

closed 2023-09-26 by kyleisah