kyleisah/Klipper-Adaptive-Meshing-Purging #186: Still Meshes the Entire Board, even with M117

raddacle, 2023-09-26:
**Describe the issue**
I'm unable to get it to follow the print area coordinates in the g-code, and instead it just meshes the entire board.

**What slicer are you using?**
Cura

**Troubleshooting**
I've triple checked the setup

The start g-code in Cura:
M117
START_PRINT BED_TEMP={material_bed_temperature_layer_0} EXTRUDER_TEMP={material_print_temperature_layer_0}

Here's a link to a copy of my config folder, and I've also included the klippy log and the gcode I've been testing it with:
[](https://drive.google.com/drive/folders/1qJYhzqz-y1CpATGUvUquXxmBk_oBfCB8?usp=sharing)

kyleisah, 2023-09-26:
Looking at your gcode file, for some reason moonraker isn't seeing that your gcode has labeled objects and is not defining them in your gcode. 

I also noticed that you have includes for KAMP purging and meshing in your printer.cfg file as well as in the KAMP settings file. You really only need to use the include lines inside of the KAMP settings file, and just include the settings file in printer.cfg

raddacle, 2023-09-26:
I appreciate you looking through it. I've cleaned up the extra includes, though I'm not sure what to do next to debug this. 

Do you have a g-code file that has worked for you before, that I could use just to test if the adaptive mesh calibration works with my setup?

kyleisah, 2023-09-26:
You can test that the adaptive meshing is working by defining an object and running the command. What is the build volume of your machine? (I've also been meaning to make some kind of test macro that shows that KAMP is working to prevent issues like this happening lol. The test macro would show you that it all works, and if your gcode isn't making it happen, then there are deeper issues outside of KAMP)

raddacle, 2023-09-26:
My bed is 300x300 (ender 3 s1 plus). Test macros would be helpful, though I know that takes a bit of work. I've just asked chatgpt to make test code, and upon running it, the printer still did a full bed calibration. Does the test code chatgpt made look right to you?

; Sample G-code for testing adaptive meshing
; Note: This is a very basic example and should be used for testing purposes only

; Initialize the printer
G28 ; Home all axes

; Define the object (using a hypothetical MESH label, as the exact format may vary)
;MESH:Test_Object

; Move to a safe position for calibration
G1 Z15.0 F9000 ; Move the platform down 15mm
G1 X0 Y0 F9000 ; Move the nozzle to the front-left corner

; Run the bed mesh calibration
BED_MESH_CALIBRATE

; Start the test print (for demonstration purposes, let's assume a 20x20x20mm cube)
G1 Z0.2 F9000 ; Move the nozzle close to the bed
G1 X20 Y20 F9000 ; Move to the start position of the cube
G92 E0 ; Reset the extruder position
G1 F140 E30 ; Extrude 30mm of filament at a speed of 140mm/min
G92 E0 ; Reset the extruder position
G1 F9000 ; Set the movement speed back to normal

; End of test G-code
M104 S0 ; Turn off the extruder
M140 S0 ; Turn off the heated bed
M84 ; Disable all stepper motors

raddacle, 2023-09-26:
I've run the sequence again, and I'm seeing that KAMP is running but its finding that the dimensions of the print area are the same as the default bounds, so it just calibrated the entire bed. Though I am unclear how it retrieved those values.

Here's KAMP's section of the log:

Default mesh bounds: (20.0, 20.0), (283.0, 263.0).
Mesh margin is 0, margin not increased.
Fuzz amount is 0, mesh points not fuzzed.
Adapted mesh bounds: (20.0, 20.0), (283.0, 263.0).
KAMP adjustments successful. Happy KAMPing!
Generating new points...
bed_mesh: generated points
Index |  Tool Adjusted  |   Probe
  0   | (51.8, 60.5)    | (20.0, 20.0)
  1   | (139.5, 60.5)   | (107.7, 20.0)
  2   | (227.1, 60.5)   | (195.3, 20.0)
  3   | (314.8, 60.5)   | (283.0, 20.0)
  4   | (314.8, 141.5)  | (283.0, 101.0)
  5   | (227.1, 141.5)  | (195.3, 101.0)
  6   | (139.5, 141.5)  | (107.7, 101.0)
  7   | (51.8, 141.5)   | (20.0, 101.0)
  8   | (51.8, 222.5)   | (20.0, 182.0)
  9   | (139.5, 222.5)  | (107.7, 182.0)
  10  | (227.1, 222.5)  | (195.3, 182.0)
  11  | (314.8, 222.5)  | (283.0, 182.0)
  12  | (314.8, 303.5)  | (283.0, 263.0)
  13  | (227.1, 303.5)  | (195.3, 263.0)
  14  | (139.5, 303.5)  | (107.7, 263.0)
  15  | (51.8, 303.5)   | (20.0, 263.0)
Updated Mesh Configuration:
relative_reference_index: None
x_count: 4
y_count: 4
mesh_x_pps: 2
mesh_y_pps: 2
algo: lagrange
tension: 0.2
Mesh Bed Leveling Complete
save_config: set [bed_mesh default] version = 1
save_config: set [bed_mesh default] points = 
  0.037500, 0.070000, 0.100000, 0.095000
  -0.052500, -0.005000, 0.045000, 0.065000
  -0.015000, 0.012500, 0.070000, 0.102500
  0.100000, 0.115000, 0.150000, 0.192500

kyleisah, 2023-09-26:
I really had to dig to find this macro, I forgot I made one a while ago that I used during early dev.

```ini
[gcode_macro _KAMP_Test_Object]
description: This macro will generate an object called "Test" in the center of your bed for testing KAMP
variable_obj_size: 50
gcode:
    {% set obj_center_x = printer.toolhead.axis_maximum.x / 2 | float %}
    {% set obj_center_y = printer.toolhead.axis_maximum.y / 2 | float %}
    EXCLUDE_OBJECT_DEFINE NAME=Debug CENTER={obj_center_x},{obj_center_y} POLYGON=[[{obj_center_x - (obj_size / 2)},{obj_center_y - (obj_size / 2)}],[{obj_center_x + (obj_size / 2)},{obj_center_y + (obj_size / 2)}]]
    { action_respond_info("Generated a {}x{}mm test object at {!r}, ready to test.".format(
        (obj_size),
        (obj_size),
        (obj_center_x, obj_center_y),
    )) }
```

Simply add this macro to your config, save and restart, and run these commands:

```ini
G28
_KAMP_TEST_OBJECT
BED_MESH_CALIBRATE
EXCLUDE_OBJECT_DEFINE RESET=1
```

That will home the machine, generate an object, (hopefully) perform an adapted mesh, and then clear the object.

Let me know what happens.

raddacle, 2023-09-27:
That worked! And so that points to the issue likely being with the slicer, Cura. I reviewed past KAMP issues and found in issue #60  someone identified that uploading/printing directly from Cura, using the Octoprint extension, skips moonrakers post-processing.
https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/issues/60#issuecomment-1526050564

After manually uploading to Mainsail and printing, adaptive meshing now works. I can't believe the issue was so simple.

kyleisah, 2023-09-27:
@raddacle fantastic, glad to hear. These things happen when you try to make something work for everyone, despite the vast differences in everyones machines lol. Enjoy! 🎉

closed 2023-09-27 by kyleisah

raddacle, 2023-09-27:
Thanks for all your help and work on this project!!