kyleisah/Klipper-Adaptive-Meshing-Purging #188: Mesh min/max calculation don't work even with exclude_object and Moonraker preprocessing

TheLexus, 2023-09-29:
I tried KAMP with my fluiddpi installation (fluid/moonraker/klipper). But it seems the bed mesh always falls back to the maximum size. I first checked if exclude_object works and it seems to work. If I download the gcode file from Fluidd it has the EXCLUDE_OBJECT stuff added, so preprocessing seems to work. I even can cancel each object separately... 

Now I tried to understand how KAMP is working. Can you explain shortly where the exclude stuff comes into play. Is it in bed_mesh_min and max?

I attached a gcode after preprocessing its done. 

**What slicer are you using?**
Ideamaker

[Gitter_Front_Rechts-2.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12771117/Gitter_Front_Rechts-2.txt)

kyleisah, 2023-09-30:
[image]

In your gcode file, EXCLUDE_OBJECT is happening *before* START_PRINT is. This needs to be the other way around. 

Objects need to be defined first, so when START_PRINT is called, and a mesh is generated, KAMP will see it and adjust the mesh. 

In your slicer, in the starting gcode section, try adding M117 before START_PRINT. This should help the preprocessor put the exclude definitions in the correct spot.

kyleisah, 2023-10-07:
Any updates on this?

TheLexus, 2023-10-07:
Hi. Had no time to verify, but I guess you are right, sounds logical. I keep you updated!

TheLexus, 2023-10-08:
OK, here is a small mod with the M117 before my START_PRINT. But I don't work that way either. I downloaded the preprocessed file and you can see, the exclude objects are after the START_PRINT. Do you have any other ideas?

[Manta-M8P_Bracket-preprocessed.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12840481/Manta-M8P_Bracket-preprocessed.txt)
[Manta-M8P_Bracket.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12840482/Manta-M8P_Bracket.txt)

kyleisah, 2023-10-08:
I really wish I knew what to tell you. It looks like you're doing everything right, but moonraker is still processing objects and putting the definitions in the wrong spot. 

By any chance, would you try a different slicer and see if that makes any difference?

TheLexus, 2023-10-11:
Ok, found what happens in case of idea maker as slicer.
Take a look at the code at https://github.com/kageurufu/preprocess_cancellation/blob/main/preprocess_cancellation.py

Here you see that the processing for idea maker differs to Cura and differs to slicer derivates. And because idea maker is adding the object information AFTER the start code stuff, it will never work. I don't know about the other because I needed to setup them first. 

At the moment I have no idea to force idea maker to something else because it's simply coded as that.

closed 2023-10-23 by kyleisah