kyleisah/Klipper-Adaptive-Meshing-Purging #190: Help with correct start_print macro...

sleewok, 2023-10-02:
**Describe the issue**
I have always used my slicer to do the start gcode. When I use the gcode in my slicer the mesh is never created at print. I have exclude objects, etc. Tried M117.

Is there a specific start_print macro that I should be using? If so, where can I find it?

**What slicer are you using?**
Orcaslicer

I appreciate any help!

sleewok, 2023-10-02:
Okay, I ended up generating a START_PRINT macro. I just started a print and it probed the print area once the bed was heated! Here is the start code:

[gcode_macro START_PRINT]
gcode:
	{% set BED_TEMP=params.BED_TEMP|default(60)|float %}
	{% set EXTRUDER_TEMP=params.EXTRUDER_TEMP|default(190)|float %}
	
	# Home the printer
	G90
	M83
	G28
	
	# Preheat the bed
	M140 S{BED_TEMP}
	M190 S{BED_TEMP}
	M104 S190
	
	# Z probing sequence
	BED_MESH_CALIBRATE
	BED_MESH_PROFILE LOAD=default
	
	# Heat the extruder to the desired temperature
	M104 S{EXTRUDER_TEMP}
	M109 S{EXTRUDER_TEMP}
	
	# Prime line sequence
	G1 Z5 F3000 ; lift
	G1 X20 Y5 F1500 ; move to prime position
	G1 Z0.15 F3000 ; get ready to prime
	G92 E0 ; reset extrusion distance
	G1 X100 E30 F600 ; prime nozzle with adjusted extrusion / E__ based on how much you like
	
	# String removal circle after priming
	G1 Z0.2 F3000 ; adjust to 0.2mm above the bed
	G1 Y15 F10000 ; move the toolhead in the Y direction by 15 units
	
	# Execute the circle 3 times
	G2 I-5 J0 F10000 ; circle with 5mm radius
	G2 I-5 J0 F10000
	G2 I-5 J0 F10000

sleewok, 2023-10-02:
It is working, but the issue I am having now is that the mesh probe count is being changed from what is specified in my printer.cfg. Why is the adapted probe count being set to 3,3?

Here is the console output:
// Algorithm: lagrange.
// Default probe count: 5,5.
// Adapted probe count: 3,3.
// Default mesh bounds: (3.0, 6.0), (179.0, 210.0).
// Mesh margin is 5.0, mesh bounds extended by 5.0mm.
// Fuzz amount is 0, mesh points not fuzzed.
// Adapted mesh bounds: (72.8316, 64.1524), (147.168, 155.848).
// KAMP adjustments successful. Happy KAMPing!
// Generating new points...

kyleisah, 2023-10-02:
I'm not sure I follow. That's the mesh size being adapted. Do you not want the mesh size to be changed?

sleewok, 2023-10-02:
> I'm not sure I follow. That's the mesh size being adapted. Do you not want the mesh size to be changed?

Originally I have set a 5,5 probe count, but it is only doing 3,3 probe count. It is probing the correct area, but it is only doing 9 points instead of 25.

kyleisah, 2023-10-02:
3,3 is the smallest mesh size that KAMP will make. Also, the probe_count in printer.cfg will be the largest mesh size that KAMP will generate.

kyleisah, 2023-10-02:
> > I'm not sure I follow. That's the mesh size being adapted. Do you not want the mesh size to be changed?
> 
> 
> 
> Originally I have set a 5,5 probe count, but it is only doing 3,3 probe count. It is probing the correct area, but it is only doing 9 points instead of 25.

Increase the probe count in your printer config to make denser meshes in smaller areas.

closed 2023-10-02 by kyleisah