kyleisah/Klipper-Adaptive-Meshing-Purging #192: VOROn_Purge code error

mcfool12, 2023-10-06:
The Voron_Purge.cfg file has an error with the line 

G1 X{purge_x_origin+size*0.289} Y{purge_y_origin+size} E{purge_amount/4} F{purge_move_speed}# Purge first line of logo

There needs to be a space for the comment to work.  It is editable but then the repo is dirty in Klipper.

kyleisah, 2023-10-06:
Just to clarify, the macro works, but because of the space there the syntax highlighting looks wrong? I can fix that easily if so

mcfool12, 2023-10-07:
It worked but would freeze after the third line, for about 15 seconds after making the purge, for me for some reason.  It started after applying the latest update, not sure if that is when it was changed.  Once I added the space it was back to working normally.

kyleisah, 2023-10-07:
That's really odd. But I will go ahead and make the changes now.

kyleisah, 2023-10-07:
There we are, changes are made and update's pushed.

closed 2023-10-07 by kyleisah