kyleisah/Klipper-Adaptive-Meshing-Purging #193: purge line not working

orochifj, 2023-10-07:
**Describe the issue**
purge line not working

**What slicer are you using?**
Orcaslicer

**Troubleshooting**
newbie not sure how to troubleshoot...
adaptive meshing is working.

**Screenshots**
[image]
[image]

**Klipper log file:**
11:41 AM
Unknown command:"PRINT_START"
11:41 AM
Bed Mesh state has been saved to profile [default]
for the current session. The SAVE_CONFIG command will
update the printer config file and restart the printer.
11:41 AM
Mesh Bed Leveling Complete
11:41 AM
probe at 166.000,165.500 is z=1.760000
 -- deleted meshing log --
1 | (86.5, 39.5) | (54.5, 24.5)
11:39 AM
0 | (60.0, 39.5) | (28.0, 24.5)
11:39 AM
bed_mesh: generated points
Index | Tool Adjusted | Probe
11:39 AM
Generating new points...
11:39 AM
KAMP adjustments successful. Happy KAMPing!
11:39 AM
Adapted mesh bounds: (28.0, 24.5), (134.0, 150.5).
11:39 AM
Fuzz amount is 0, mesh points not fuzzed.
11:39 AM
Mesh margin is 10.0, mesh bounds extended by 10.0mm.
11:39 AM
Default mesh bounds: (5.0, 10.0), (134.0, 160.0).

**Sliced Gcode**
[flowrate_0_PETG_32m25s.txt](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12836741/flowrate_0_PETG_32m25s.txt)

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-10-07:
Looks like you don't have a print_start macro

kyleisah, 2023-10-07:
I would remove that print_start macro line you have there and just add `LINE_PURGE` to the very end of your machine's print start gcode.

I would also highly recommend asking around and learning a bit about macros and how they work, and find a good starter print_start gcode macro. A good place to start is the Klipper discord.

closed 2023-10-07 by kyleisah