kyleisah/Klipper-Adaptive-Meshing-Purging #195: Purge Line always start at (0,0)

ylqiu2000, 2023-10-12:
**Describe the issue**
I have Voron 0 without any probes, the Purge Line always starts (0,0) regardless of Line_Purge or Voron_Purge. 

**What slicer are you using?**
Prusaslicer

**Troubleshooting**
KAMP_Setting is responding correctly after changing the parameters inside the file. I have tried with different objects and locations of the build plate. Exclude object is working correctly as it shows up during print. 

**Screenshots**
[image]
[image]
[image]

**Klipper log file:**
[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12885675/klippy.log)

kyleisah, 2023-10-12:
In your slicer, try adding M117 before print_start. It's a common issue with the preprocessor unfortunately.

ylqiu2000, 2023-10-12:
That is such a niche error, thank you very much for resolving this!

closed 2023-10-12 by ylqiu2000