kyleisah/Klipper-Adaptive-Meshing-Purging #196: error when trying to do bed mesh

VHumpal, 2023-10-13:
**Describe the issue**
A clear and concise description of what the bug is...
i start my print it heats up homes and then i get this error when it trys to doing the probing 

Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _KAMP_Settings'

**What slicer are you using?**
Prusaslicer, Superslicer, Cura, Orcaslicer...
im using cure 5.4 and i have tried orca slicer

**Troubleshooting**
What troubleshooting steps have you tried?
i have tried diffrent slicers chnaging settings 
**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Klipper log file:**
Please add klipper log file here...

[klippy.log.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12895242/klippy.log.zip)

**Sliced Gcode**
Please add sliced gcode file you are using here...

[CE3S1PRO_3dbenchy.gcode.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12895231/CE3S1PRO_3dbenchy.gcode.zip)

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-10-13:
At a glance I think you forgot to [include] KAMP_Settings.cfg

VHumpal, 2023-10-13:
where do i need to include it?

kyleisah, 2023-10-13:
In printer.cfg

VHumpal, 2023-10-13:
i got it working thank you

kyleisah, 2023-10-13:
No worries!

closed 2023-10-13 by kyleisah