kyleisah/Klipper-Adaptive-Meshing-Purging #199: wont do purge line

VHumpal, 2023-10-13:
**Describe the issue**
A clear and concise description of what the bug is...
i just got it set up but now i cant get my purge line to run it just immedietly goes to the skirt of my 3d print 
**What slicer are you using?**
Prusaslicer, Superslicer, Cura, Orcaslicer...
orca slicer
**Troubleshooting**
What troubleshooting steps have you tried?
none
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...
[klippy.log.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12899277/klippy.log.zip)

**Sliced Gcode**
Please add sliced gcode file you are using here...
[3dbenchy_PLA_1h18m.gcode.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/12899280/3dbenchy_PLA_1h18m.gcode.zip)

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-10-14:
Looks like you don't have a PRINT_START macro or something. Log is full of messages about it. Give me a moment

kyleisah, 2023-10-14:
You should replace this section in orcaslicer:
[image]
With this:
```ini
M117
PRINT_START EXTRUDER_TEMP=[first_layer_temperature] BED_TEMP=[first_layer_bed_temperature]
```

and then replace your START_PRINT macro with PRINT_START (it's more commonly referred to as PRINT_START):

```ini
[gcode_macro PRINT_START]
gcode:
    {% set BED_TEMP = params.BED_TEMP|default(110)|int %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(240)|int %}

    M190 S{BED_TEMP}            ; wait for bed temp to stabilize
    G28                         ; Home all
    BED_MESH_CALIBRATE
    SMART_PARK
    M109 S{EXTRUDER_TEMP}
    LINE_PURGE
```

Keep in mind that PRINT_START is meant to be tailored to your machine specifically, so feel free to customize it. As it is right now, this macro will pull the temps from the gcode file, warm the bed, home the machine, make a mesh, park, heat up the nozzle, purge, and finally the print will happen. I hope this helps!

VHumpal, 2023-10-14:
it still wouldnt do it i got this 

[Extruder] max_extrude_cross_section is insufficient for purge, please set it to 5 or greater. Purge skipped.

kyleisah, 2023-10-14:
Right, this is covered in the readme. 

Here's a link: https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#adaptive-purging

It's at the bottom, in the note.

VHumpal, 2023-10-14:
oh sorry must have missed that

kyleisah, 2023-10-15:
@VHumpal No worries, I know it's kind of a lot all at one time lol. Did that get you up and running?

VHumpal, 2023-10-15:
everything is working perfectly, know its super nice having a level bed on every print

closed 2023-10-15 by kyleisah