kyleisah/Klipper-Adaptive-Meshing-Purging #200: to much max velocity

labotecno, 2023-10-15:
**Describe the issue**
hello 
after the bedmesh i get to much max velocity to park the toolhead , and fallow with purge 

**What slicer are you using?**
 Orcaslicer

**Screenshots**
https://www.youtube.com/watch?v=iJJnZ7mRcNE

**Klipper log file:**
`[gcode_macro PRINT_START]
gcode:
#------Get Variables-----#
  {% set BED = params.BED|default(65)|float %}
  {% set BED_TEMP = params.BED_TEMP|default(0)|float %}
  {% set EXTRUDER = params.EXTRUDER|default(210)|float %}
  {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(0)|float %}
  {% set MATERIAL = params.MATERIAL|default("material") %}

  {% if (EXTRUDER_TEMP > 0) %}
    {% set EXTRUDER = EXTRUDER_TEMP %}
  {% endif %}

  {% if (BED_TEMP > 0) %}
    {% set BED = BED_TEMP %}
  {% endif %}

#-----Initialize-----#
  RESPOND MSG="Initilizing..."
  G21          ; set to mm
  M220 S100	 ; set print speed to 100%
  M221 S100	 ; set flow rate to 100%
  M107         ; disable fans
  G92 E0       ; set extruder to zero

#-----Home and Clear Bed mesh-----#
  BED_MESH_CLEAR
  STATUS_HOMING
  M104 S170 ; set extruder temp for bed leveling
  M140 S{BED_TEMP} ; set bed temp
  M109 S170 ; wait for bed leveling temp
  M190 S{BED_TEMP} ; wait for bed temp
  G28
  G90          ; absolute positioning
  M83           ; extruder to relative mode

#-----Heat soak Bed----#
  RESPOND MSG="Preheating..."
  STATUS_BED_HEATING
  G1 Z20 F500                ; move nozzle away from bed
  SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET={BED}    ; set bed temp
    #{% if BED > 80 %}
      #SET_FAN_SPEED FAN=Nevermore SPEED=0.5
      #M106 S200
      #SET_HEATER_TEMPERATURE HEATER=extruder TARGET=50         ; set extruder temp
    #{% endif %}
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={BED}         ; wait for bed temp
    
#------Heat soak chamber if requested-----#
#If the bed is set above 95 we can assume that we are using the printer enclosed and therefore should SOAK the chamber
  #{% if BED > 95 %}                            ; Turn on bed and nevermore fans for ABS and set to 100% until the chamber is at temp
    #RESPOND MSG="Heating Chamber..."
      #{% if BED <= 110 %}    
        #SET_FAN_SPEED FAN=Nevermore SPEED=.75
    	#TEMPERATURE_WAIT SENSOR="temperature_sensor chamber_temp" MINIMUM=45  
      #{% endif %}
      #{% if BED > 110 %}
        #SET_FAN_SPEED FAN=Nevermore SPEED=.75
    	#TEMPERATURE_WAIT SENSOR="temperature_sensor chamber_temp" MINIMUM=55  
      #{% endif %}
  #{% endif %}

#------Preheat nozzle-----#
    {% if MATERIAL == "PLA" %}
      SET_HEATER_TEMPERATURE HEATER=extruder TARGET=120        ; set extruder temp to 120 for QGL w/o oozing (PLA)
    {% else %}
      SET_HEATER_TEMPERATURE HEATER=extruder TARGET=150        ; set extruder temp to 160 for QGL w/o oozing
    {% endif %}
    STOP_LED_EFFECTS

#------Quad Gauntry level-----#
  LIGHTS_ON
  RESPOND MSG="z tilt Gantry Leveling..."
  STATUS_LEVELING
  Z_TILT_ADJUST
  STOP_LED_EFFECTS

#-----Bed Mesh-----#
  RESPOND MSG="Bed mesh measurement..."
  STATUS_MESHING
  BED_MESH_CALIBRATE
  #BED_MESH_PROFILE LOAD=default                           ; load default mesh

#-----Final Heat-----#
  #M106 S0
  Smart_Park
  RESPOND MSG="Waiting for hotend..."
  STATUS_PRINTING
  CASELIGHTS_ON
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET={EXTRUDER}
  TEMPERATURE_WAIT SENSOR=extruder MINIMUM={EXTRUDER}      ; wait for extruder temp

#-----Print-----#
  STATUS_PRINTING
    {% if BED >= 95 %}
      {% if MATERIAL == 'PC' %}
         SET_FAN_SPEED FAN=Nevermore SPEED=0.25
      {% else %}
        SET_FAN_SPEED FAN=Nevermore SPEED=1
      {% endif %}
    {% endif %}
  RESPOND MSG="Start printing !"
  STATUS_PRINTING
  {% if MATERIAL == 'PLA' %}
        Line_Purge
      {% else %}
        VORON_PURGE
  {% endif %}
#  SFS_ENABLE
  M117`

**Sliced Gcode**

**Additional context**
Add any other context about the problem here.

thank you to help me and you make really good stuff with KAMP!

kyleisah, 2023-10-16:
Can you elaborate a bit on this? Is the move too rapid for your machine, and you skip steps?

If that's the case then you have your `max_velocity` value under `[printer]` too high. 

The intent of `max_velocity` is the maximum velocity your machine is capable of (travel speed). 

Lower this value to your desired travel speed and your machine will never exceed that speed under normal operating circumstances.

kyleisah, 2023-10-16:
Oh you have a video, that's new for me haha. 

Yeah, looks like just an error in your config. Adjust the value mentioned above and you won't have this issue anymore.

labotecno, 2023-10-16:
i have max_velocity at : max_velocity: 300
i will try little more . 
can you modificate this in your code 
`{% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %}                                                           # Set travel speed from config`
for printer really fast it can be a problem . 
i try to modificate , but its only in read only 
thank you

kyleisah, 2023-10-16:
> i have max_velocity at : max_velocity: 300 i will try little more . can you modificate this in your code `{% set travel_speed = (printer.toolhead.max_velocity) * 60 | float %} # Set travel speed [...]

No, I will not modify that value. That is the intended formula for printer movements. You need to either LOWER `max_velocity`, or increase your XY motor currents. 

My `max_velocity` is 600 on my machine because my machine can handle the speeds.

labotecno, 2023-10-16:
i put to 150 just to try it  and dont breaking anything , its working fine , i will increase this value little by little  , thank you to the support. why  the code is read only  ?

closed 2023-10-18 by labotecno