kyleisah/Klipper-Adaptive-Meshing-Purging #202: Move out of range

hemna, 2023-10-16:
**Describe the issue**
A clear and concise description of what the bug is...
I just installed KAMP on my klipper install following the instructions on the README on github.
When I try and print the klipper adaptive purge tower, the line purge throws the Move out of range.

I didn't know where else to ask for help.

```09:11:47 
echo: KAMP Line Purge
09:11:47 
// Moving filament tip 0.0mms
09:11:47 
// KAMP purge is not using firmware retraction, it is recommended to configure it.
09:11:47 
// KAMP purge starting at 102.5, 64.334 and purging 30.0mm of filament, requested flow rate is 12.0mm/s3.
09:11:47 
!! Move out of range: 102.500 64.334 -1.343 [0.000]
09:11:47 
!! Move out of range: 102.500 64.334 -1.343 [0.000]
09:11:47 
!! Move out of range: 102.500 64.334 -1.343 [0.000]
```

**What slicer are you using?**
Cura 

**Troubleshooting**
No idea how to troubleshoot this.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...

**Sliced Gcode**
Please add sliced gcode file you are using here...

**Additional context**
Add any other context about the problem here.

```
Moving filament tip 0.0mms
KAMP purge is using firmware retraction.
KAMP purge starting at 102.5, 64.334 and purging 30.0mm of filament, requested flow rate is 12.0mm/s3.
Move out of range: 102.500 64.334 -1.347 [0.000]
Move out of range: 102.500 64.334 -1.347 [0.000]
Move out of range: 102.500 64.334 -1.347 [0.000]
SDCARD_RESET_FILE cannot be run from the sdcard
SDCARD_RESET_FILE cannot be run from the sdcard
virtual_sdcard on_error
Traceback (most recent call last):
  File "/home/waboring/klipper/klippy/extras/virtual_sdcard.py", line 264, in work_handler
    self.gcode.run_script(line)
  File "/home/waboring/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/waboring/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/waboring/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/waboring/klipper/klippy/extras/gcode_macro.py", line 189, in cmd
    self.template.run_gcode_from_command(kwparams)
  File "/home/waboring/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/waboring/klipper/klippy/gcode.py", line 213, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/waboring/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/waboring/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
[... 425 lines cut ...]
G0 F6000 X78.497 Y77.904 Z0.2
;TYPE:SKIRT
G11
G1 F1200 X79.476 Y77.069 E0.0428
G1 X80.138 Y76.568 E0.07041
G1 X80.673 Y76.185 E0.09229
G1 X81.466 Y75.681 E0.12354
G1 X82.373 Y75.211 E0.15752
G1 X83.353 Y74.762 E0.19337
G1 X84.19 Y74.422 E0.22342
G1 X85.407 Y74.062 E0.26563
G1 X86.17 Y73.884 E0.29169
G1 X86.746 Y73.764 E0.31126
G1 X87.751 Y73.611 E0.34507
G1 X88.693 Y73.55 E0.37647
G1 X89.495 Y73.534 E0.40315
G1 X90.163 Y73.539 E0.42537
G1 X91.135 Y73.599 E0.45776
G1 X91.997 Y73.727 E0.48674
G1 X93.197 Y73.964 E0.52743
G1 X94.502 Y74.261 E0.57194
G1 X97 Y74.793 E0.65689
...
...
```

kyleisah, 2023-10-16:
What do you have the purge_height variable set to?

hemna, 2023-10-17:
variable_purge_height: 0.8

Here is my KAMP_Settings.cfg

```
# Below you can include specific configuration files depending on what you want KAMP to do:

[include ./KAMP/Adaptive_Meshing.cfg]       # Include to enable adaptive meshing configuration.
[include ./KAMP/Line_Purge.cfg]             # Include to enable adaptive line purging configuration.
# [include ./KAMP/Voron_Purge.cfg]            # Include to enable adaptive Voron logo purging configuration.
# [include ./KAMP/Smart_Park.cfg]             # Include to enable the Smart Park function, which parks the printhead near the print area for final heating.

[gcode_macro _KAMP_Settings]
description: This macro contains all adjustable settings for KAMP 

# The following variables are settings for KAMP as a whole.
variable_verbose_enable: True               # Set to True to enable KAMP information output when running. This is useful for debugging.

# The following variables are for adjusting adaptive mesh settings for KAMP.
variable_mesh_margin: 20                     # Expands the mesh size in millimeters if desired. Leave at 0 to disable.
variable_fuzz_amount: 5                     # Slightly randomizes mesh points to spread out wear from nozzle-based probes. Leave at 0 to disable.

# The following variables are for those with a dockable probe like Klicky, Euclid, etc.                 # ----------------  Attach Macro | Detach Macro
variable_probe_dock_enable: False           # Set to True to enable the usage of a dockable probe.      # ---------------------------------------------
variable_attach_macro: 'Attach_Probe'       # The macro that is used to attach the probe.               # Klicky Probe:   'Attach_Probe' | 'Dock_Probe'
variable_detach_macro: 'Dock_Probe'         # The macro that is used to store the probe.                # Euclid Probe:   'Deploy_Probe' | 'Stow_Probe'
                                                                                                        # Legacy Gcode:   'M401'         | 'M402'

# The following variables are for adjusting adaptive purge settings for KAMP.
variable_purge_height: 0.8                  # Z position of nozzle during purge, default is 0.8.
variable_tip_distance: 0                    # Distance between tip of filament and nozzle before purge. Should be similar to PRINT_END final retract amount.
variable_purge_margin: 10                   # Distance the purge will be in front of the print area, default is 10.
variable_purge_amount: 30                   # Amount of filament to be purged prior to printing.
variable_flow_rate: 12                      # Flow rate of purge in mm3/s. Default is 12.

# The following variables are for adjusting the Smart Park feature for KAMP, which will park the printhead near the print area at a specified height.
variable_smart_park_height: 10              # Z position for Smart Park, default is 10.

gcode: # Gcode section left intentionally blank. Do not disturb.

    {action_respond_info(" Running the KAMP_Settings macro does nothing, it is only used for storing KAMP settings. ")}
```

kyleisah, 2023-10-17:
Well, for some reason when trying to purge, it's trying to move to a negative Z position. I think this has something to do with your bed mesh settings and not KAMP. 

Something you can try is adding `zero_reference_position` to your [bed_mesh] config block. The position should be the wherever your printer homes to if it homes with a probe, like the center point of the bed. 

More info here:

https://www.klipper3d.org/Bed_Mesh.html#configuring-the-zero-reference-position

JurajMik, 2023-10-20:
Here is sanity check for bed_mesh_calibrate.cfg

closed 2023-10-20 by kyleisah