kyleisah/Klipper-Adaptive-Meshing-Purging #203: Getting an error after enabling KAMP with purge

itssunnyb, 2023-10-21:
After i enabled KAMP with purge line. I get this error **Move out of range: 128.109 28.354 -0.082 [30.000]**. It does the bed mesh correctly and prints the purge line, but i get that error. If i disable KAMP i can print normally.

kyleisah, 2023-10-21:
Can you upload a klipper log and a gcode file for me?

itssunnyb, 2023-10-21:
just tried to print this file but getting the same error Move out of range: 52.931 96.671 -0.159 [27.000] i can upload a g code file
[klippy (2).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/13059816/klippy.2.log)

kyleisah, 2023-10-21:
I think what might be going on is you have `position_min: 0` under `[stepper_z]` and the nozzle is trying to move to a negative Z position, but your config will not allow it.

I try to leave a little padding in my machine configs for thicker buildplates and stuff like that, so I usually have `position_min` for Z at something like `-2`. 

Give that a shot.

itssunnyb, 2023-10-21:
> I think what might be going on is you have `position_min: 0` under `[stepper_z]` and the nozzle is trying to move to a negative Z position, but your config will not allow it.
>
> I try to leave a [...]

That did it, thank you!!

kyleisah, 2023-10-21:
Great to hear. Happy KAMPing :)

closed 2023-10-21 by kyleisah