kyleisah/Klipper-Adaptive-Meshing-Purging #205: Move speed too fast

OrbitalExplorer, 2023-10-21:
**Describe the issue**
Printer tries to move too fast when going to purge/smart park position. This results in the move being too fast and the printer unable to complete it and just skips steps

**What slicer are you using?**
Prusaslicer

**Troubleshooting:**
Removing the * 60 at the speed parameter fixes it, I think

**Possible Solution:**
Add the speed in the Kamp_settings.cfg and pull from there instead of mulitiplying by 60

kyleisah, 2023-10-21:
Hello! I see this often and the reason is almost always a misunderstanding in machine config. What do you have for `max_velocity` under [printer] in printer.cfg?

OrbitalExplorer, 2023-10-21:
I have a speed of 1300 mm/s there. I thought that since it said "max_speed," I needed to check my maximum possible speed, and it turned out to be 1300 without skipping.

If it's something else what should I set it to? (I print at 110-180 mm/s)

kyleisah, 2023-10-21:
You should have that set to whatever you use for your travel speed when printing. That value is pretty much the maximum feedrate in mm/s that you want your machine to move at. 1300mm/s is… pretty fast lol. 

200-300mm/s is a good value for most. Set it to 200 and give it a shot.

OrbitalExplorer, 2023-10-23:
Sorry for the delay in getting back to you. I'll give it a shot, but won't it still be too quick since 200 multiplied by 60 is still 12,000?

kyleisah, 2023-10-23:
What you're talking about is the feedrate conversion. 

In klipper configs, you're defining the speed in millimeters per second, but in gcode, the value needs to be millimeters per *minute*, which is why that conversion is there.

OrbitalExplorer, 2023-10-23:
Oh now I see, thanks for the help.

closed 2023-10-23 by OrbitalExplorer