kyleisah/Klipper-Adaptive-Meshing-Purging #209: I've just installed kamp but it's not doing line purge

drumstick77, 2023-10-27:
Hi All,

I've been chatting with some guys on different discord servers, and some of them are experiencing the same issue where there the line is not purging.
I've got it working on ender 3 S1 Pro which I've installed a few months back...and it's still working
but I've just installed it for the first time on Ender 3 V2 Neo and Ender 5 Pro, and both are not working
the bed mesh calibration is working though, it's just not doing the line purging.

just wondering if anyone else is having similar issue? Thanks

kyleisah, 2023-10-28:
If I knew what the issue was I would be able to help. 

Can you tell me what the console says when you are trying to purge?

drumstick77, 2023-10-28:
Thank you.
It's this error 
[image]

where can I find the settings for that again?
thanks

drumstick77, 2023-10-28:
forgot to add that under extruder within printer.cfg

closed 2023-10-28 by drumstick77