kyleisah/Klipper-Adaptive-Meshing-Purging #210: adaptive meshing in the wrong position

rafaelmontserrat, 2023-10-29:
**Describe the issue**
The adaptive meshing is happening in a different spot (looks like it has an offset in the positive X direction) than the real printing spot.
**What slicer are you using?**
Orcaslicer

**Troubleshooting**
I have tried adjusting my position min and position max in my printer.cfg file but that has not solved the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Klipper log file:**
Please add klipper log file here...
I do not know how to extract this from my klipper interface (mainsail)

**Sliced Gcode**
Please add sliced gcode file you are using here...
[trngls tght - Part 1_PLA_3h28m.gcode.zip](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/13198878/trngls.tght.-.Part.1_PLA_3h28m.gcode.zip)

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-10-29:
I think you need to dial in your config to get 0,0 to be the very front left corner of your bed. 

Without those being known correct, you're going to have all kinds of weird issues, outside of KAMP as well. 

A friend of mine has a helpful guide to getting the origin (0,0) point of your bed perfect:

https://github.com/rootiest/zippy_guides/blob/main/guides/axis_limits.md

Get that figured out and adjust your config, then try it again. As long as 0,0 in your slicer and 0,0 on your machine are at the same position, wherever you have the print located in your slicer will be in the expected location on your bed.

rafaelmontserrat, 2023-10-30:
Hey. Thanks for the suggestion, but something is still off. My slicer is set on 0,0 as the printer is (u cant see the toolhead in the pic but u can see the shadow, its on the left front). After following that guide, I've got my 0,0 at the front left in the printer and a bunch of negative values in the printer.cfg since my end stops are in the completely opposite corner (back right). Anyways, I sliced a 25x25mm cube in the middle of my bed and when the KAMP triggered it started to probe one of the sides of the bed, rather than the middle, so something its still off and I can't figure out what. Any ideas? Maybe some probe offset or some other offsets in the slicer? As I mentioned at the beginning the Y coordinate of the probing looks ok but the X looks way too close (with this setup) to X=0 rather than X=115 (my bed is approx 230x225).

[image]
[image]

kyleisah, 2023-10-31:
Hm… sounds more like maybe it’s your probe offsets now possibly. 

If the probe sits BEHIND the nozzle, the Y offset is positive. If the probe sits RIGHT of the nozzle, the X offset is positive. 

Sounds like maybe you’ve got one or both of those offsets opposite what they need to be but I can’t be sure.

rafaelmontserrat, 2023-10-31:
Just double and triple checked... my probe offsets are correct... :(

rafaelmontserrat, 2023-10-31:
Hm... My probe sits in front of the nozzle and on the left, so both my offsets are negative. However, I have swapped my X offset to be positive and now the adaptive meshing seems to be working fine, so I don't know what happened there but I guess problem solved! Thanks for the help mate, great job on this macros.

closed 2023-11-01 by kyleisah