kyleisah/Klipper-Adaptive-Meshing-Purging #211: KAMP triggering MCU Panic

Eliminateur, 2023-10-31:
**Describe the issue**
Running BED_MESH_CALIBRATE after several points crashes klipper with "**MCU 'mcu' shutdown: ADC out of range**"
On small real prints(that only probe the surface of the object) it doesn't happens, seems to dependant on how much time probing takes

Probe is BLtouch, XY cartesian printer.

**Troubleshooting**
Ran BED_MESH_CALIBRATE from the klipper console on a completely cold printer (no heaters running)

In the attached klippy log on line 6708 you can see the last received command before the crash:
07:46:11 PM Probing failed due to printer shutdown
07:46:11 PM Klipper state: Shutdown
07:46:09 PM probe at 273.000,62.831 is z=1.135625

Already updated all the software to the latest available
[image]

[klippy.log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/13210689/klippy.log)

kyleisah, 2023-10-31:
ADC being out of range typically relates to a thermistor issue. Like a wire breakage, loose connection, dying thermistor etc. 

I uploaded your log file to an analyzer and got your temperature graph here:

[image]

So it looks like this isn’t a KAMP issue, but a machine issue.

closed 2023-10-31 by kyleisah

Eliminateur, 2023-11-01:
¿which klipper log analyzer are you using?, i can't see those negative temp excursions in mainsail graphs...

kyleisah, 2023-11-02:
I'm using [Klippylyzer](https://klippylyzer.github.io/)