kyleisah/Klipper-Adaptive-Meshing-Purging #215: Any way to bypass the "No objects detected" error when using Creality Print?

kr0mka, 2023-11-01:
When sending prints through Creality Print with calibration checked on, I'm getting the "No objects detected! Check your gcode and make sure that EXCLUDE_OBJECT_DEFINE is happening before BED_MESH_CALIBRATE is called. Defaulting to regular meshing." log message and then the printer does a regular full bed mesh. Same thing happens when starting a print through the screen with calibration enabled, without calibration enabled the bed leveling is being skipped (as expected I assume). I have the "exclude objects" enabled in Creality Print. I've checked my gcode and this is how the object definitions look: 

"; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 2 known objects
EXCLUDE_OBJECT_DEFINE NAME=0_0 CENTER=110,110 POLYGON=[[14.225000,72.581000],[14.225000,147.419000],[205.775000,147.419000],[205.775000,72.581000],[14.225000,72.581000]]
EXCLUDE_OBJECT_DEFINE NAME=0_1 CENTER=119.79,168.67 POLYGON=[[75.134000,149.670000],[75.134000,187.670000],[164.446000,187.670000],[164.446000,149.670000],[75.134000,149.670000]]"

Not sure what's wrong here, i'd like to keep using the one-click printing option in creality with kamp meshing working properly. Anything I'm missing?

kyleisah, 2023-11-02:
A klipper log file and a sliced gcode file from mainsail/fluidd would be really helpful. Without those, I can't see anything.

kyleisah, 2023-11-02:
You need to make sure those object definition commands are happening BEFORE bed_mesh_calibrate is being called, as the error states

kyleisah, 2023-11-05:
Closing due to inactivity

closed 2023-11-05 by kyleisah