kyleisah/Klipper-Adaptive-Meshing-Purging #217: Adaptive Meshing not obeying probe offset

stephenmott, 2023-11-02:
In my printers probes config I have an offset of -44 specified for the y value i.e.

[probe]
pin: !gpio22
x_offset: 0
y_offset: -44
speed: 5

So the probe, as I look at the printer is 44mm to the left of the nozzle,  but when I use Adaptive Meshing it seems to ignore this and probe an area 44mm to the left of where the print is.

Not sure if this is a bug, or if I've just missed something..?

kyleisah, 2023-11-02:
If the probe sits BEHIND the nozzle, the Y offset is positive. If the probe sits RIGHT of the nozzle, the X offset is positive.

If the probe is left of the nozzle, shouldn’t you have a negative **X** offset?

stephenmott, 2023-11-02:
I don't think so, the 0,0 is front right of the printer, y is left to right, x is front to back (it's an original Cetus3D, but I've replaced most of the internals and given it a lead screw), the probe actually hangs off the front of the nozzle, but I've rotated the whole assembly to get more print area. E.g. if I go 
G0 X100 Y100
Note the position of the nozzle, then
G0 X100 Y56
The probe is on top of the previous point.

kyleisah, 2023-11-02:
Well, there's 2 things to this now. 

1, KAMP does not touch your printers probe offsets. If you had a working bed mesh configuration before, all KAMP does is shrink the mesh bounds to the objects passed from your slicer. 

2, I don't think you have your machine configured correctly. Klipper defines the bed origin (0,0) as the **front left** corner of square/rectangular beds. 

That can be seen when looking at the bed mesh documentation images [here](https://www.klipper3d.org/Bed_Mesh.html#basic-configuration). 

Also, X movement has always been left and right, and Y movement has always been front and back. 

Until you've got your machine configured correctly for klipper, you're going to have a lot of strange problems like these.

closed 2023-11-02 by kyleisah

stephenmott, 2023-11-02:
I mean if it does a normal mesh it probes the entire bed, well almost, the nozzle can only go 20mm outside the bed surface in y  and with a 44mm offset on the proble there's 20mm it can't reach.

Surely 0,0 is relative, if I rotate my whole printer 90º then 0,0 is bottom left with x left / right and y front / back. If you did that then at 0,0 the proble would be at x=0, y=44. I did mess about a lot to get it working, but if made it +44 it just wouldn't cover the entire bed.