kyleisah/Klipper-Adaptive-Meshing-Purging #218: Request to move the Purge Line

Merexor, 2023-11-03:
The "problem" is that my fan is quite close to the nozzle and with a thicker purge line the fan takes it with it because it is made under the printed image. If it were printed over it you would drive away without the fan having to go over the purge line.

And I'm thinking of some kind of setting value in the kamp_settings that you can use to determine the location.

kyleisah, 2023-11-03:
You can try reducing purge height and purge amount in KAMP_Settings to make the purge line less tall

kyleisah, 2023-11-03:
You can also set the purge margin to a very large number and the purge will be moved to the left side of the bed and purge along Y rather than X

Merexor, 2023-11-03:
Verstehe ich das richtig das die Linie unten links gedruckt wird ?
Wie wird das berechnet und könnte man hier nicht irgendwie dann die Koordinaten ändern ?

like:
purge_line_destination = 0 (default)

0= links unten ( +x / + y )
1= rechts unten ( -x / +y )
2= links oben ( +x / - y )
3= rechts oben ( -x / -y )

kyleisah, 2023-11-03:
I think you would be better off not using KAMP purging if you want it in a specific area. 

Find a purge to add to your slicer gcode and use that instead

closed 2023-11-03 by kyleisah

Merexor, 2023-11-05:
Can you explain to me what the individual variables stand for or refer to?

purge_x_min 
purge_y_min 
purge_x_max
purge_y_max 
purge_x_center 
purge_y_center 
purge_x_origin 
purge_y_origin