kyleisah/Klipper-Adaptive-Meshing-Purging #219: Issue BED_MESH_CALIBRATE

MrCrashy, 2023-11-03:
After starting the Print follow Issue is shown in the Console:
`Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': jinja2.exceptions.UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'gcode_macro _KAMP_Settings'`

I tried with Cura und PrusaSlicer, I tried with M117 before my START_PRINT. I checked the KAMP_Settings too. I cant find the Problem.

kyleisah, 2023-11-03:
You don't have KAMP_Settings included in your printer.cfg

MrCrashy, 2023-11-03:
Ok, its working. Dont know why, but the printer.cfg didnt save. 
Now I have following Issue:

No objects detected! Check your gcode and make sure that EXCLUDE_OBJECT_DEFINE is happening before BED_MESH_CALIBRATE is called. Defaulting to regular meshing.

I Enabled the "Object Lable"-Settings and did the [exclude_object].

kyleisah, 2023-11-03:
Try the first step listed here:

https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging#troubleshooting

That's a good way to force those definitions to happen in the right spot.

kyleisah, 2023-11-03:
Oh you've already tried that...

kyleisah, 2023-11-03:
Have you added the sxclude object stuff to moonraker.conf?

The enable_object_processing thing?

If you did, you have to restart moonraker too. 

Might be best to just reboot your pi altogether to make sure all those changes take effect.

MrCrashy, 2023-11-03:
Like i said... Im very new in Klipper. After Editing all .cfgs i only took "Firmware restart". I dont knew, that i have to restart Moonraker too. 
Moonraker shows me KAMP "dirty" what coud I do?

kyleisah, 2023-11-06:
If it's getting marked dirty you are editing the files inside of the KAMP folder rather than editing the KAMP_settings file that was copied into your config directory.