kyleisah/Klipper-Adaptive-Meshing-Purging #220: Adaptive Mesh working but not probing the front of the bed.....

Fearlessyota, 2023-11-05:
**Describe the issue**
When I start a print, it will probe everywhere it needs to but the front portion of the print area. It will also not probe the rear of the bed due to the offset of the probe. So when slicing, I know I need to leave space toward the rear of the bed. But still it doesn't probe the front portion.

**What slicer are you using?**
Superslicer
**Troubleshooting**
What troubleshooting steps have you tried?
I have adjusted the probe offset multiple times. I may still be getting it wrong, but how it behaves probing the rear portion of the bed seem spot on. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Klipper log file:**
Please add klipper log file here...
[klippy (2).log](https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/files/13258099/klippy.2.log)

**Sliced Gcode**
Please add sliced gcode file you are using here...
G28 ; home all axes

BED_MESH_CALIBRATE

LINE_PURGE

**Additional context**
Add any other context about the problem here.

kyleisah, 2023-11-05:
I'm not sure I understand. You want the mesh to adapt, right?

I don't see any errors in your logic file. The points are generated and are within the original mesh bounds.

Fearlessyota, 2023-11-05:
[image]

Attached is a screenshot, In this example it will only probe within the red circle. It misses the portion toward the front of the bed. Sometimes this give me 1st layer issues in that area of the bed. The areas it does probe the first layer is perfect every time

If I center the piece to the middle of  the bed I will get to an error when it tries to probe beyond the limit of the Y axis. This is why its positioned on the bed as you see it.

kyleisah, 2023-11-06:
Sounds like your config is not correct, or your slicer profile does not match your machine config.

Fearlessyota, 2023-11-06:
Hmmm, I will take another look. I thought I had the kinks worked out of the configs. Thanks for taking a look at this